jensrossbach/esphome-ferraris-meter #53: High measurement from flickering when rotation/consumtion is very low

michnixweiss, 2025-01-08:
**Is your change request related to a problem? Please describe.**
I have a ferraris meter with 75u/kWh. The problem is when power consumtion is low <200 W the sensor is sometimes flickering at the crossover. Which results in a high consumtion measurement.

**Describe the solution you'd like**
Is it possible to implement various debounce thresholds depending on rotation or consumtion? Eg 2 sec <200W, 400ms >=200W

**Describe alternatives you've considered**
Leave it as it is and manually correct the aberration in HA.

**Additional context**
Never th less a hugh thanks for.this peace of software!

gramels, 2025-01-09:
Did you use the digital or the analog version? I do have 75 r/kWh and do not get flickering at low wattage. For the digital version you nee do play with the poti and find the point where it just turns off. This is a fiddely exercise.

jensrossbach, 2025-01-09:
> Is it possible to implement various debounce thresholds depending on rotation or consumtion? Eg 2 sec <200W, 400ms >=200W

I don't think that this is possible as the consumption value is the result of a calculation based on the rotation time. So you would have a kind of chicken-egg-problem. Theoretically, the average consumption from some recent time period could be used but to calculate this, it is necessary to keep a history of past consumption values and this costs memory which is quite limited on the ESP microcontrollers. Also I doubt that this will help much.

Have you already tried to increase the value for `debounce_threshold`? This should definitely help. You can also try to use the analog output of the infrared sensor as you have a bit more control over the signal in this case and you do not need to fiddle around with the potentiometer. Additionally, you should review if you can optimize the placement and alignment of the infrared sensor. From my experience, this can significantly increase the reliability of the signal.

> In case it persist a possible approach could be to introduce a max wattage value and everything above is considered flickering.

Increasing the value of `debounce_threshold` basically has the same effect as if you would limit the power consumption value (the only difference is that the power consumption is based on the time between two rising edges while the debounce threshold is based on the time between a falling and the subsequent rising edge, but the difference should be neglectable),

gramels, 2025-01-11:
today I got funny values as well
[image]

michnixweiss, 2025-01-11:
Thanks for the very detailed answer! This will help me estimate it better.
Since we don't have an electric car, we should never reach this consumption range.
I'll try 800ms ;-)

gramels, 2025-01-11:
Could backwards counting trigger such unrealistic values? Eg when PV Energy Production is higher than consumption?

michnixweiss, 2025-03-01:
So I got no more peaks and a divergence of less than 1kWh per week - what is good enough for me.
I close this as solved.
Thanks!

closed 2025-03-01 by michnixweiss