jensrossbach/esphome-ferraris-meter #54: Stuck in Boot Loop on ESP32C3 (Seeed XIAO ESP32C3)

adriaanb, 2025-02-12:
**Describe the bug**
Microcontroller ends up in safe mode soon after flashing due to a boot loop. After the initial flashing process I was able to add it to HomeAssistant, but ever since power cycling it after that initial flash, nothing helps to resolve the issue. It boot loops and gets stuck in safe mode, HA cannot reach it.

**To reproduce**
Steps to reproduce the behavior:
1. Flash with the attached config
2. Observe Serial Logs

**Expected behavior**
The device should boot successfully and remain stable without entering a boot loop or safe mode.

**Logs and configuration**

*Firmware config*
```yaml
esphome:
  name: stromzaehler
  friendly_name: Stromzähler

esp32:
  board: seeed_xiao_esp32c3
  framework:
    type: arduino

external_components:
  - source: github://jensrossbach/esphome-ferraris-meter@v1.4.1
    components: [ferraris]

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxx"

wifi:
  ssid: "xxxxxxxxxxxxxxxxxx"
  password: "xxxxxxxxxxxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Stromzaehler Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxx"

captive_portal:

ferraris:
  id: ferraris_meter
  digital_input: GPIO10
  rotations_per_kwh: 375
  debounce_threshold: 300
  energy_start_value: last_energy_value
    
# numeric sensors
sensor:
  - platform: ferraris
    # sensor for current power consumption
    power_consumption:
      name: Momentanverbrauch
    # sensor for energy meter reading
    energy_meter:
      name: Verbrauchszähler
  # sensor for Wi-Fi signal strength
  - platform: wifi_signal
    name: Wi-Fi Signal
    update_interval: 60s

# binary sensors
binary_sensor:
  - platform: ferraris
    # sensor for indicating detected rotation during calibration mode
    rotation_indicator:
      name: Umdrehungsindikator
  # sensor for wireless connection status
  - platform: status
    name: Status

# text sensors
text_sensor:
  - platform: wifi_info
    # sensor providing the IP address of the ESP microcontroller
    ip_address:
      name: IP Adresse

# number components
number:
  # import entity holding last energy meter reading from Home Assistant
  - platform: homeassistant
    id: last_energy_value
    entity_id: input_number.stromzaehler_letzter_wert
  # number used to manually overwrite the energy meter
  - platform: template
    id: target_energy_value
    name: Manueller Zählerstand
    icon: mdi:counter
    unit_of_measurement: kWh
    device_class: energy
    entity_category: config
    mode: box
    optimistic: true
    min_value: 0
    max_value: 1000000
    step: 0.01

# switch components
switch:
  - platform: ferraris
    # switch used to activate or deactivate the calibration mode
    calibration_mode:
      name: Kalibrierungsmodus

# button components
button:
  # button used to trigger the manual overwriting of the energy meter
  - platform: template
    name: Verbrauchszähler überschreiben
    icon: mdi:download
    entity_category: config
    on_press:
      - ferraris.set_energy_meter:
          id: ferraris_meter
          value: !lambda |-
            float val = id(target_energy_value).state;
            return (val >= 0) ? val : 0;
```

*Serial logs*

```log
[19:51:50]Build:Feb  7 2021
[19:51:50]rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[19:51:50]Saved PC:0x40381a90
WARNING Decoded 0x40381a90: esp_restart_noos_dig at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:44 (discriminator 1)
[19:51:50]SPIWP:0xee
[19:51:50]mode:DIO, clock div:1
[19:51:50]load:0x3fcd6100,len:0x438
[19:51:50]load:0x403ce000,len:0x918
[19:51:50]load:0x403d0000,len:0x24e4
[19:51:50]entry 0x403ce000
[19:51:50][I][logger:171]: Log initialized
[19:51:50][C][safe_mode:079]: There have been 9 suspected unsuccessful boot attempts
[19:51:50][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:51:50][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[19:51:50][I][app:029]: Running through setup()...
[19:51:50][D][number:012]: 'Manueller Zählerstand': Sending state 0.000000
[19:51:51][D][sensor:094]: 'Momentanverbrauch': Sending state 0.00000 W with 1ESP-ROM:esp32c3-api1-20210207
[19:51:51]Build:Feb  7 2021
[19:51:51]rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[19:51:51]Saved PC:0x40381a90
WARNING Decoded 0x40381a90: esp_restart_noos_dig at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:44 (discriminator 1)
[19:51:51]SPIWP:0xee
[19:51:51]mode:DIO, clock div:1
[19:51:51]load:0x3fcd6100,len:0x438
[19:51:51]load:0x403ce000,len:0x918
[19:51:51]load:0x403d0000,len:0x24e4
[19:51:51]entry 0x403ce000
[19:51:51][I][logger:171]: Log initialized
[19:51:51][C][safe_mode:079]: There have been 10 suspected unsuccessful boot attempts
[19:51:51][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:51:51][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[19:51:51][E][safe_mode:086]: Boot loop detected. Proceeding to safe mode
[19:51:51][E][component:164]: Component safe_mode set Error flag: unspecified
```

**Context information**
 - ESPHome version: 2024.12.2
 - Ferraris Meter version: Actaris (manufacturer), 375U/kWh
 - Microcontroller type: Seeed XIAO ESP32C3

**Additional Information**

- The microcontroller works normally using a minimal configuration without the ferraris component.
- The device was tested with USB power and separate 5V power.

jensrossbach, 2025-02-13:
I don't have a spare ESP32 microcontroller at the moment but I will try to reproduce the problem on an ESP8266 as soon as I have some time. If I'm not able to reproduce the issue, I will come back to you and probably need your support to debug the problem.

adriaanb, 2025-02-24:
I very much appreciate that! Once it manages to boot it does run smoothly (although I see the behavior described in issue #53 which I will make a separate post about after some more observation.) That said, any reset or power cycle significantly impacts the smoothness of the user experience. I’m happy to assist with the debugging process in any way I can.

One thought that occurred to me is that the ESP32-C3, unlike the "standard" ESP32 or ESP8266, is based on a RISC-V architecture rather than a Xtensa-based architecture. I assume Espressif has taken steps to maintain consistency within the product family, so I am not sure if this is a relevant bit of context.

Let me know if there’s any specific debugging information or testing that would be helpful!

jensrossbach, 2025-02-24:
Have you already seen/tried this?
https://community.home-assistant.io/t/seeed-xiao-esp32c3-not-usable-so-far-with-esphome/543596/12

jensrossbach, 2025-02-26:
Ok, meanwhile I was also able to test your configuration (with slight modifications of course in order to fit to an ESP8266) on a D1 mini but I cannot reproduce the issue. Also my "production" setup at my Ferraris electricity meter works very well (also a D1 mini).

adriaanb, 2025-03-12:
So, I have been able to test a few things, but unfortunately to no avail. I was not able to add a delat at the start of the ferraris component initialization, but I tried the following modification to my esphome block:

```yaml
esphome:
  name: stromzaehler
  friendly_name: Stromzähler

  # Delay
  on_boot:
    priority:  700
    then:
      - delay: 5s
```

I did, however, notice something. The boot loop occurs after attempting to send a state, and not always for the same sensor. Once the board ends up performing a successful boot, I would, for instance, see:

```
[11:23:25][D][binary_sensor:036]: 'Umdrehungsindikator': Sending state OFF
```

Until this happens, however, I see this right before the boot loop issue manifests:

```
[15:44:24][D][binary_sensor:036]: 'Umdrehungsindikator': Sending stESP-ROM:esp32c3-api1-20210207
```

immediately followed by a reboot:

```
[15:44:24]Build:Feb  7 2021
[15:44:24]rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
[15:44:24]Saved PC:0x40381a90
WARNING Decoded 0x40381a90: esp_restart_noos_dig at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:44 (discriminator 1)
```

The string `ESP-ROM:esp32c3-api1-20210207`appears right before every instance of the boot loop that I could identify. Here an example in the middle of logging the sending of another state:

```
[17:17:43][D][sensor:094]: 'Momentanverbrauch': Sending state 0.000ESP-ROM:esp32c3-api1-20210207
```

 Here again the section of the log right after flashing the firmware:

```
Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbmodem2101 with baud rate 115200
[15:44:23][I][logger:171]: Log initialized
[15:44:23][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[15:44:23][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:44:23][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[15:44:23][I][app:029]: Running through setup()...
[15:44:23][D][number:012]: 'Manueller Zählerstand': Sending state 0.000000
[15:44:23][D][sensor:094]: 'Momentanverbrauch': Sending state 0.00000 W with 1 decimals of accuracy
[15:44:24][D][binary_sensor:036]: 'Umdrehungsindikator': Sending stESP-ROM:esp32c3-api1-20210207
[15:44:24]Build:Feb  7 2021
[15:44:24]rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
[15:44:24]Saved PC:0x40381a90
WARNING Decoded 0x40381a90: esp_restart_noos_dig at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:44 (discriminator 1)
[15:44:24]SPIWP:0xee
[15:44:24]mode:DIO, clock div:1
[15:44:24]load:0x3fcd6100,len:0x438
[15:44:24]load:0x403ce000,len:0x918
[15:44:24]load:0x403d0000,len:0x24e4
[15:44:24]entry 0x403ce000
[15:44:24][I][logger:171]: Log initialized
[15:44:24][C][safe_mode:079]: There have been 1 suspected unsuccessful boot attempts
[15:44:24][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:44:24][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[15:44:24][I][app:029]: Running through setup()...
[15:44:24][D][number:012]: 'Manueller Zählerstand': Sending state 0.000000
[15:44:24][D][sensor:094]: 'Momentanverbrauch': Sending state 0.00000 W with 1 decimals of accuracy
[15:44:24][D][binary_sensor:036]: 'Umdrehungsindikator': Sending stESP-ROM:esp32c3-api1-20210207
[15:44:24]Build:Feb  7 2021
[15:44:24]rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
[15:44:24]Saved PC:0x40381a90
.
.
.
etc...
```

Not sure if any of this holds useful information of any kind. In any case, safe mode is disabled after 5 minutes, and thus in most cases the boot loop begins again, until the rare case occurs that it just begins working...

Can I somehow hard-code a value for every component to determine if that makes a difference? Alternatively, can you suggest a more targeted way of introducing a delay before initializing the ferraris component without blocking the remainder of esphome processes?

jensrossbach, 2025-03-12:
I assume the string `ESP-ROM:esp32c3-api1-20210207` is the very first output of every boot cycle and therefore comes on every instance of the boot loop. The system seems to crash in the middle of logging the sensors published value. But I have no clue what could cause this.

Anyway, I prepared a change that delays the setup function for 5 seconds and adds some additional instrumentation to the setup function. You can try this out by changing the `external_components` block like follows:

```yaml
external_components:
  - source: github://jensrossbach/esphome-ferraris-meter@debugging/54
    components: [ferraris]
```

adriaanb, 2025-03-13:
Thank you very much for this!

I tried it, but unfortunately, the same behavior keeps occuring, just at a later stage. It even gets to a "working" state briefly, registering state changes to the sensor if they occur before the crash. It now also allows me to briefly get logs via WiFi as well before the crash occurs, which previously worked only via Serial. 

Bottom line, it seems to have postponed the problem slightly. It still "recovers" to a working state eventually, and I have no idea why or how. I disassembled and rewired everything in the meantime, tried different pins, and got it to a state where it seemed to work for a while... just for it to begin exhibiting the issue again after 3 or 4 reboots.

What I find odd is that OTA updates using `esphome run` always work, regardless of whether or not I am able to connect to it using `esphome logs`. I don't know what else I could sensibly try at this point. If there is anything I can do to contribute to debugging this, please let me know. Otherwise I may just have to go ahead and oder an ESP8266.

adriaanb, 2025-03-13:
This morning I looked again at the logs, here is the additional information produced thanks to the changes you've provided:

_Logs using debug branch, digital sensor, and arduino core:_
```
[12:08:33][D][ferraris:080]: Setup START
[12:08:33][D][ferraris:104]: Initializing m_power_consumption_sensor
[12:08:33][D][sensor:094]: 'Momentanverbrauch': Sending state 0.00000 W with 1 decimals of accuracy
[12:08:33][D][ferraris:112]: Initializing m_rotation_indicator_sensor
[12:08:33][D][binary_sensor:036]: 'Umdrehungsindikator': Sending state OFF
[12:08:34][D][ferraris:162]: Setting up m_debounce_threshold_numberESP-ROM:esp32c3-api1-20210207
[12:08:34]Build:Feb  7 2021
[12:08:34]rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[12:08:34]Saved PC:0x40381a90
WARNING Decoded 0x40381a90: esp_restart_noos_dig at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:44 (discriminator 1)
```

I also remembered there is an alternative core for the esp32c3 provided by espressif. It has proven finnicky in the past, but I had hoped it might solve any issues related to the arduino core in this case. Replacing `type: arduino` with `type: esp-idf` 

 ```yaml
esp32:
  board: seeed_xiao_esp32c3
  framework:
    type: esp-idf
```

did not solve the issue, but provided possibly more helpful error logs. The crashes still occur in the same location, but now it returns this:

_Logs using main branch, digital sensor, and espressif core:_
```
[11:49:32][I][app:029]: Running through setup()...
[11:49:32][D][number:012]: 'Manueller Zählerstand': Sending state 0.000000
[11:49:32][D][sensor:093]: 'Momentanverbrauch': Sending state 0.00000 W with 1 decimals of accuracy
[11:49:32][D][binary_sensor:036]: 'Umdrehungsindikator': Sending statGuru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
[11:49:32]
[11:49:32]Core  0 register dump:
[11:49:32]MEPC    : 0x4200ac60  RA      : 0x4200aba2  SP      : 0x3fca0610  GP      : 0x3fc92000
WARNING Decoded 0x4200ac60: esphome::ferraris::FerrarisMeter::setup() at /Users/adrib/.esphome/build/stromzaehler/src/esphome/components/ferraris/ferraris_meter.cpp:151
WARNING Decoded 0x4200ac60: esphome::ferraris::FerrarisMeter::setup() at /Users/adrib/.esphome/build/stromzaehler/src/esphome/components/ferraris/ferraris_meter.cpp:151
```

_Logs using debug branch, digital sensor, and espressif core:_
```
[12:12:23][D][ferraris:080]: Setup START
[12:12:23][D][ferraris:104]: Initializing m_power_consumption_sensor
[12:12:23][D][sensor:093]: 'Momentanverbrauch': Sending state 0.00000 W with 1 decimals of accuracy
[12:12:23][D][ferraris:112]: Initializing m_rotation_indicator_sensor
[12:12:23][D][binary_sensor:036]: 'Umdrehungsindikator': Sending state OFF
[12:12:23][D][ferraris:162]: Setting up m_debounce_threshold_numberGuru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
[12:12:23]
[12:12:23]Core  0 register dump:
[12:12:23]MEPC    : 0x4200ade8  RA      : 0x4200ade0  SP      : 0x3fca05d0  GP      : 0x3fc92000
WARNING Decoded 0x4200ade8: std::_Function_handler<void (), esphome::ferraris::FerrarisMeter::setup()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /Users/adrib/.esphome/build/stromzaehler/src/esphome/components/ferraris/ferraris_meter.cpp:163
 (inlined by) __invoke_impl<void, esphome::ferraris::FerrarisMeter::setup()::<lambda()>&> at /Users/adrib/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/12.2.0/bits/invoke.h:61
 (inlined by) __invoke_r<void, esphome::ferraris::FerrarisMeter::setup()::<lambda()>&> at /Users/adrib/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/12.2.0/bits/invoke.h:111
 (inlined by) _M_invoke at /Users/adrib/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/12.2.0/bits/std_function.h:290
WARNING Decoded 0x4200ade8: std::_Function_handler<void (), esphome::ferraris::FerrarisMeter::setup()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /Users/adrib/.esphome/build/stromzaehler/src/esphome/components/ferraris/ferraris_meter.cpp:163
 (inlined by) __invoke_impl<void, esphome::ferraris::FerrarisMeter::setup()::<lambda()>&> at /Users/adrib/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/12.2.0/bits/invoke.h:61
 (inlined by) __invoke_r<void, esphome::ferraris::FerrarisMeter::setup()::<lambda()>&> at /Users/adrib/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/12.2.0/bits/invoke.h:111
 (inlined by) _M_invoke at /Users/adrib/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/12.2.0/bits/std_function.h:290
```

_Logs using debug branch, analog sensor, and espressif core:_
```
[11:35:42][D][sensor:093]: 'adc_input': Sending state 515.00000 V with 2 decimals of accuracy
[11:35:42][D][sensor:093]: 'adc_input': Sending state 515.00000 V with 2 decimals of accuracy
[11:35:42][D][ferraris:080]: Setup START
[11:35:42][D][ferraris:084]: Setting up m_analog_input_sensor
[11:35:42][D][ferraris:104]: Initializing m_power_consumption_sensor
[11:35:42][D][sensor:093]: 'Momentanverbrauch': Sending state 0.00000 W with 1 decimals of accuracy
[11:35:42][D][ferraris:112]: Initializing m_rotation_indicator_sensor
[... 29 lines cut ...]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
[... cut ...]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
Compiling .pioenvs/stromzaehler/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/stromzaehler/src/esphome/components/binary_sensor/filter.cpp.o
```

This is at tthe specified location in ferraris_meter.cpp:

```cpp
if (m_debounce_threshold_number->has_state())
{
    m_debounce_threshold = m_debounce_threshold_number->state;
}
```

Please let me know if there is any other thing I can do to help!

jensrossbach, 2025-03-15:
> This morning I looked again at the logs, here is the additional information produced thanks to the changes you've provided

To be honest, I don't think that I can do much about this in the Ferraris component. The component doesn't do anything special or significantly different than other ESPHome components. The code lines where the problem happens do not show any obvious faults and, depending on the timing, the problem seems to happen at different places. The error "Load access fault" indicates a problem with memory access which is probably caused by lower layers of the firmware or drivers (maybe unaligned memory access or something similar). 

Have you tried to omit the `framework` configuration under the  `esp32` block? Maybe this leads to ESPHome chosing the most appropriate framework for the hardware itself. But that's just an idea, don't know if this helps. Maybe you can also try to use "esp32-c3-devkitm-1" for the `board` configuration as a variant from the discussion thread I posted above.

> Likely unrelated, but while testing compiling the analog sensor config, it also produced deprecation warnings during the compilation process

Yes, I think this is unrelated and I didn't observe this when compiling the code for the ESP8266 platform. Anyway, thanks for the hint.

ospringauf, 2026-01-03:
I had the same issue on an ESP32-C3 "Supermini" board. Let Copilot have a look at the problem - it seems that a member variable had not been initialized, this caused the crash during boot.
I have opened a PR for this. Thanks for this project!

closed 2026-01-04 by jensrossbach