jensrossbach/esphome-ferraris-meter #55: Use with digital power meter

michnixweiss, 2025-03-01:
It's not a real bug cause I try to use it for a non ferraris meter.
Nevertheless I hope for help.

Context Info:
ESPHome 2025.2.1
D1 mini ESP8266
  with four TCRT5000 as digital input
  at three ferraris meter
  and now one digital, which is the actual problem.

I got a new power meter WS7612.2 with a IR check led which gives 10.000 imp/kWh. As per description it has 50% pulse duty factor and 2ms pulse length.

So I disabled the emitting diode and put the receiver in front of the pulse led of the WS7612.2
Configured with
rotations_per_kwh: 5000
debounce_threshold: 0

Basically its working, but the measurement is way low.

Now I don't know:
If in generel I can "misuse" this project for this powermeter.
If my config is correct.
If there are too much signals and my d1 mini cannot process all correctly. 
- Possible solution: only send every 10th signal!?

Thanks in advance!

jensrossbach, 2025-03-01:
I think such use case is not in scope of the Ferraris meter component. I propose that you maybe try out the pulse counter and/or pulse meter ESPHome built-in components. They are probably better suited for such high frequent, even pulses.

Anyway, a problem report issue is certainly the wrong means for such a topic, better use the discussion forum for this. For example, you can convert this issue into a discussion.

michnixweiss, 2025-03-06:
Hy @jensrossbach,
thanks for the answer, i will go give the pulse counter a try - cause it updates in a specified time period and not at each pulse.

As i wrote "it's not a real bug", but i cannnot find anything to transform it to a discussion.