jensrossbach/esphome-ferraris-meter #61: High values on HA (re)start

michnixweiss, 2025-04-10:
**Describe the bug**
Sometimes I have a very high energy value on one or two of my 4 energymeter running on a ESP8266.
Digging into it i found out that the values are the current ferrarismeter values and the moment they occure is the start of HA after a shutdown/reboot.
So for some reason the ferrarismeter drop to zero and then syncs the correct value, which HA interprets in a high consumption... maybe!?

**To reproduce**
Shutdown/reboot HA, mainly when shuting down the hole VM.

**Context information**
 - HA version: 2025.3.4
 - ESPHome version: 2025.3.3
 - Ferraris Meter version: 1.4.1
 - Microcontroller type (e.g., ESP32, ESP8266): ESP8266

The peaks are only shown in the statistics, not in the sensor history.
I added some screenshots.

[image]
[image]
[image]

jensrossbach, 2025-04-16:
How do you know that the value of the sensor drops to 0 if it is not visible in the history? From the second picture above, all data delivered by the sensor looks fine. Also, I have never seen such behavior in my setup at home (and I regularly restart Home Assistant for updates).

As I unfortunately do not have the time for trying to reproduce this on my side, can you please activate debug logs and capture them while the problem occurs on your side? Please attach the logs to this issue, thanks!

michnixweiss, 2025-04-17:
I do not know that the sensor drops to 0, its a assumption.
Do you want the debug logs from HA or ESP?

Thanks for your efforts.

michnixweiss, 2025-04-21:
sorry, noob...
i changed logger level to debug, the peak appeared again tonight, i downloaded the logs from ESPhome Builder, but it shows only entry till i opened the logs. Is there a "history" or how do i get the logs in time of HA start/restart?
```yaml
INFO ESPHome 2025.4.0
INFO Reading configuration /config/esphome/smartmeter.yaml...
INFO Starting log output from xxx.xxx.xxx.xxx using esphome API
INFO Successfully connected to smartmeter @ xxx.xxx.xxx.xxx in 0.008s
INFO Successful handshake with smartmeter @ xxx.xxx.xxx.xxx in 2.626s
[10:04:32][I][app:100]: ESPHome version 2025.4.0 compiled on Apr 20 2025, 08:59:20
[10:04:32][C][wifi:600]: WiFi:
[10:04:32][C][wifi:428]:   Local MAC: xxxxxx
[10:04:32][C][wifi:433]:   SSID: 'xxx'[redacted]
[10:04:32][C][wifi:436]:   IP Address: xxx.xxx.xxx.xxx
[10:04:32][C][wifi:439]:   BSSID: xxxxxx[redacted]
[10:04:32][C][wifi:441]:   Hostname: 'xxx'
[10:04:32][C][wifi:443]:   Signal strength: -65 dB ▂▄▆█
[10:04:32][C][wifi:447]:   Channel: 6
[10:04:32][C][wifi:448]:   Subnet: 255.255.255.0
[10:04:32][C][wifi:449]:   Gateway: xxx.xxx.xxx.xxx
[10:04:32][C][wifi:450]:   DNS1: 8.8.8.8
[10:04:32][C][wifi:451]:   DNS2: 0.0.0.0
[10:04:32][C][logger:177]: Logger:
[10:04:32][C][logger:178]:   Max Level: DEBUG
[10:04:32][C][logger:179]:   Initial Level: DEBUG
[10:04:32][C][logger:181]:   Log Baud Rate: 115200
[10:04:32][C][logger:182]:   Hardware UART: UART0
[10:04:32][C][template.number:050]: Template Number 'Manueller Zählerstand 1'
[10:04:32][C][template.number:050]:   Icon: 'mdi:counter'
[10:04:32][C][template.number:050]:   Unit of Measurement: 'kWh'
[10:04:32][C][template.number:050]:   Device Class: 'energy'
[10:04:32][C][template.number:051]:   Optimistic: YES
[10:04:32][C][template.number:052]:   Update Interval: 60.0s
[10:04:32][C][template.number:050]: Template Number 'Manueller Zählerstand 2'
[10:04:32][C][template.number:050]:   Icon: 'mdi:counter'
[10:04:32][C][template.number:050]:   Unit of Measurement: 'kWh'
[10:04:32][C][template.number:050]:   Device Class: 'energy'
[10:04:32][C][template.number:051]:   Optimistic: YES
[10:04:32][C][template.number:052]:   Update Interval: 60.0s
[10:04:32][C][template.number:050]: Template Number 'Manueller Zählerstand 3'
[10:04:32][C][template.number:050]:   Icon: 'mdi:counter'
[10:04:32][C][template.number:050]:   Unit of Measurement: 'kWh'
[10:04:32][C][template.number:050]:   Device Class: 'energy'
[10:04:32][C][template.number:051]:   Optimistic: YES
[10:04:32][C][template.number:052]:   Update Interval: 60.0s
[10:04:32][C][template.number:050]: Template Number 'Manueller Zählerstand 4'
[10:04:32][C][template.number:050]:   Icon: 'mdi:counter'
[10:04:32][C][template.number:050]:   Unit of Measurement: 'kWh'
[10:04:32][C][template.number:050]:   Device Class: 'energy'
[10:04:32][C][template.number:051]:   Optimistic: YES
[10:04:32][C][template.number:052]:   Update Interval: 60.0s
[10:04:32][C][ferraris:212]: Ferraris Meter
[10:04:32][C][ferraris:213]:   Digital input pin: GPIO5
[10:04:32][C][ferraris:243]:   Rotations per kWh: 75
[10:04:32][C][ferraris:247]:   Static debounce threshold: 800 ms
[10:04:32][C][ferraris:253]: Power consumption sensor 'Power Consumption 1'
[10:04:32][C][ferraris:253]:   Device Class: 'power'
[10:04:32][C][ferraris:253]:   State Class: 'measurement'
[10:04:32][C][ferraris:253]:   Unit of Measurement: 'W'
[10:04:32][C][ferraris:253]:   Accuracy Decimals: 1
[10:04:32][C][ferraris:253]:   Icon: 'mdi:lightning-bolt'
[10:04:32][C][ferraris:254]: Energy meter sensor 'Energy Meter 1'
[10:04:32][C][ferraris:254]:   Device Class: 'energy'
[10:04:32][C][ferraris:254]:   State Class: 'total_increasing'
[10:04:32][C][ferraris:254]:   Unit of Measurement: 'Wh'
[10:04:32][C][ferraris:254]:   Accuracy Decimals: 1
[10:04:32][C][ferraris:254]:   Icon: 'mdi:transmission-tower'
[10:04:32][C][ferraris:257]: Rotation indicator sensor 'Umdrehungsindikator 1'
[10:04:32][C][ferraris:068]: Calibration mode switch 'Kalibrierungsmodus 1'
[10:04:32][C][ferraris:070]:   Icon: 'mdi:wrench-cog'
[10:04:32][C][ferraris:079]:   Device Class: 'switch'
[... 62 lines cut ...]
[10:04:32][C][esphome.ota:078]:   Password configured
[10:04:32][C][safe_mode:018]: Safe Mode:
[10:04:32][C][safe_mode:019]:   Boot considered successful after 60 seconds
[10:04:32][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:04:32][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[10:04:32][C][api:140]: API Server:
[10:04:32][C][api:141]:   Address: smartmeter.local:6053
[10:04:32][C][api:143]:   Using noise encryption: YES
[10:04:32][C][wifi_signal.sensor:010]: WiFi Signal 'Wi-Fi Signal'
[10:04:32][C][wifi_signal.sensor:010]:   Device Class: 'signal_strength'
[10:04:32][C][wifi_signal.sensor:010]:   State Class: 'measurement'
[10:04:32][C][wifi_signal.sensor:010]:   Unit of Measurement: 'dBm'
[10:04:33][C][wifi_signal.sensor:010]:   Accuracy Decimals: 0
[10:04:33][C][wifi_info:010]: WifiInfo IPAddress 'IP Adresse'
[10:04:33][C][homeassistant.number:072]: Homeassistant Number 'last_energy_value_1'
[10:04:33][C][homeassistant.number:073]:   Entity ID: 'input_number.energymeter_1_last_value'
[10:04:33][C][homeassistant.number:072]: Homeassistant Number 'last_energy_value_2'
[10:04:33][C][homeassistant.number:073]:   Entity ID: 'input_number.energymeter_2_last_value'
[10:04:33][C][homeassistant.number:072]: Homeassistant Number 'last_energy_value_3'
[10:04:33][C][homeassistant.number:073]:   Entity ID: 'input_number.energymeter_3_last_value'
[10:04:33][C][homeassistant.number:072]: Homeassistant Number 'last_energy_value_4'
[10:04:33][C][homeassistant.number:073]:   Entity ID: 'input_number.energymeter_4_last_value'
[10:04:35][D][ferraris:268]: State change:  0 -> 1
[10:04:35][D][ferraris:301]: Rotation time:  39693 ms
```

jensrossbach, 2025-04-21:
The only way I know is to have the logging connection active at the moment when the problem happens.

Alternatively, I found this tutorial how to create a file logger that writes the logs to a file stored on the Home Assistant server:
https://community.home-assistant.io/t/tutorial-esphome-file-logger-works-with-haos/784357

michnixweiss, 2025-04-22:
Is it possible to have a logging connection outside of HA?

jensrossbach, 2025-04-26:
Yes, you can use the ESPHome [command line tools](https://www.esphome.io/guides/getting_started_command_line) to capture the logs from a command shell.

If you have installed the tools on your computer, you can run:
```
esphome logs <your_config>.yaml
```
This way, you can also easily redirect the logs into a file. Downside is that you have to run the computer all the time while capturing the logs.

michnixweiss, 2025-07-07:
This is driving me crazy. If something's going wrong, like today (I forgot to turn off power saving mode), the error occurs. When I record the log, nothing happens.
Hopefully, I'll get it soon...

michnixweiss, 2025-12-07:
This is from my SmartMeter.yaml:
```
automation:
- id: 'nupdate_smartmeter_cache_2'
  alias: Aktualisierung Verbrauchszähler-Cache 2
  trigger:
  - platform: state
    entity_id:
      - sensor.smartmeter_energy_meter_2
  condition: []
  action:
  - action: input_number.set_value
    target:
      entity_id: input_number.energymeter_2_last_value
    data:
      value: '{{ states(trigger.entity_id) }}'
  mode: single
```
could it be a problem, that i added it to the configuration.yaml as package?
```
homeassistant:
  packages:
    smartmeter: !include xtensions/SmartMeter.yaml
```

jensrossbach, 2025-12-14:
According to ESPHome documentation of [API component](https://www.esphome.io/components/api), the ESP does a reboot if no connection to Home Assistant can be established within 15 minutes (default, see `reboot_timeout`). But I'm not sure if this applies only to the first connection after booting or if it also happens after a connection loss at any time later. Anyway, I think a Home Assistant update shouldn't take such long.